markwal/OctoPrint-GPX #89: Something something something, steps/mm

thebeline, 2021-02-24:
See #74 where I hijacked a thread.  My b.

Issue is I seem to be having some issues with `Machine Definition` not persisting or updating `steps/mm` when re-printing my test cube.

I uploaded a 20x20x20 cube in `gcode`, found my steps were off, so I adjusted the Machine Definition and re-printed, they were still the same off.

I said to heck with it, HALVED the Y-Axis, reprinted, still the same.

So I moved on, uploaded a new STL, went to print, and wouldn't you know it, the Y-Axis was halved...

So I deleted and re-uploaded the cube, printed, Y was halved.  At this moment I am fixing the `steps/mm` back to the values I had calculated, and I am gong to re-print the cube.  If they are still halved, I will delete and re-upload and see what happens.

BRB

thebeline, 2021-02-24:
Re-Printing the same file did not show my changes, however, rebooting the Pi (the whole system) and re-printing did.

It occurs to me now that the time my changes (the bad ones) were reflected, was when I had to restart OctoPi due to #88.

markwal, 2021-02-24:
So, if I understand correctly:

Steps:
* Modify machine settings
* Print

Result:
Machine settings not reflected. However:

Steps:
* Modify machine settings
* Restart octoprint (or perhaps, disconnect and reconnect the printer on the Connections box)
* Print

Result:
Machine settings are reflected

If so, then I think what is happening is that machine definition settings are only picked up when the printer is connected.

thebeline, 2021-02-25:
Hmmm...  Sounds about right.  Maybe trigger a disconnect after Machine Definition is saved?  Force a reconnect?  Or use the feature OctoPrint does when it says "Settings require a restart", I mean, it is fairly common, it would not ruffle my feathers at all.