markwal/OctoPrint-GPX #97: MakerBot Replicator - 5th Gen

beadon, 2023-03-27:
So, what do we all think of supporting a Makerbot 5th gen ?  Is it tricky to get compatible ?

beadon, 2023-04-02:
Just tearing apart a simple file.  Looks like it's just a zipfile.

Downloads % unzip M_fixedbc3ba655ae257278d4464702050f21ecb87ea1edrepairedrepaired.makerbot
Archive:  M_fixedbc3ba655ae257278d4464702050f21ecb87ea1edrepairedrepaired.makerbot
  inflating: meta.json               
  inflating: print.jsontoolpath      
  inflating: isometric_thumbnail_120x120.png  
  inflating: isometric_thumbnail_320x320.png  
  inflating: isometric_thumbnail_640x640.png  
  inflating: thumbnail_110x80.png    
  inflating: thumbnail_320x200.png   
  inflating: thumbnail_55x40.png     

Attaching the meta.json to this issue for further digging:

`
{
   "bot_type": "replicator_5",
   "bounding_box": 
   {
      "x_max": 36.35638427734375,
      "x_min": -36.35638427734375,
      "y_max": 40.51321411132812,
      "y_min": -40.51321411132812,
      "z_max": 10.99999999999998,
      "z_min": 0.20
   },
   "commanded_duration_s": 2516.320,
   "duration_s": 2516.320,
   "extruder_temperature": 215,
   "extruder_temperatures": [ 215 ],
   "extrusion_distance_mm": 1481.177436735376,
   "extrusion_distances_mm": [ 1481.177436735376 ],
   "extrusion_mass_g": 4.519237518310547,
   "extrusion_masses_g": [ 4.519237518310547 ],
   "grue_git_commit_hash": "d00e003",
   "grue_version": "7.14.0",
   "libthing_git_commit_hash": "298ec81",
   "libthing_version": "5.27.0",
   "material": "pla",
   "materials": [ "pla" ],
   "miracle_config": 
   {
      "_bot": "replicator_5",
      "_extruders": [ "mk13" ],
      "_materials": [ "pla" ],
      "configPath": "/tmp/df62d6f4-f925-4e84-a203-29b999671307/profile.json",
      "doRaft": false,
      "gaggles": 
      {
         "instance0": 
         {
            "_baseLayer": "paddedBase",
            "_printMode": "draft",
            "_supportType": "none",
            "adjacentFillLeakyConnections": true,
            "adjacentFillLeakyDistanceRatio": 1.40,
            "areaToTaperSupportsIn": 9223372036854775807,
            "areaToTaperSupportsOut": 0.0,
            "backlashEpsilon": 0.050,
            "backlashFeedback": 0.90,
            "backlashX": 0.0,
            "backlashY": 0.090,
            "baseLayer": "paddedBase",
            "baseLayerHeight": 0.20,
            "bedZOffset": 0.0,
            "bridgeAnchorMinimumLength": 0.80,
            "bridgeAnchorWidth": 0.80,
            "bridgeMaximumLength": 80.0,
            "bridgeRasterCellPerMM": 8.0,
[... 1327 lines cut ...]
      }
   ],
   "num_tool_changes": 0,
   "num_z_layers": 109,
   "num_z_transitions": 108,
   "platform_temperature": 0,
   "preferences": 
   {
      "instance0": 
      {
         "baseLayer": "paddedBase",
         "layerHeight": 0.10,
         "modelFillProfiles.sparse.density": 12,
         "printMode": "draft"
      }
   },
   "sliceconfig_git_commit_hash": "9379c8a",
   "sliceconfig_version": "9.14.0",
   "tool_type": "mk13",
   "tool_types": [ "mk13" ],
   "total_commands": 65127,
   "uuid": "c7ba4422-d2e8-4c79-a298-64f7b28402f8",
   "version": "1.2.0"
}
`

charely6, 2024-04-16:
so I'm not part of this project, but from what I understand the makerbot replicator 5th gen is really closed source and different than the earlier makerbot replicators. You should look at the mbotmake repos here on github. Its a python script to convert a gcode file into a makerbot file like the one you opened there. There are some forks and it needs to be cleaned up some but using the fork I made and tweaked on my makerbot mini+ has been working pretty good. (Note I'm sending the files using the makerbot print legacy program and controlling it through that.)

beadon, 2024-04-17:
Ok, I am going to close this issue. I have done my research now.  There are better , more advanced , more compatible 3d printers that are on the market for very little money.  Reverse engineering MakerBot protocols and files formats - at this point -  given the EoL status of the parts and the broken nature of the software seems like a fools errand.

I will close this issue.

closed 2024-04-17 by beadon