markwal/OctoPrint-GPX #99: Errorno 5 can't connect to replicator mini+

wBrhy2, 2024-06-25:
Hello, I've been trying to fix some replicator mini + for my local library and the software is so locked down that I can't even get the printers to manually heat to clear a clog, so I set up an old mbp as a Ubuntu computer with octoprint.

I have tried all baud rates and default back to 115200 when trying anything
Serial port set to auto scans through all 32 ttyS devices

All I get is

Unexpected error while connecting to serial port /dev/ttyS#, baud rate 115200 from hook GPX: OSerror '[errorno 5] input/output error' dev/ttyS#'' @ comm.py:_open_serial:3944

Where # is all ttyS from 0 to 31 sequentially, all with the same error.

lsusb does list the makerbot device.

wBrhy2, 2024-06-25:
In verbose logs I get

Unable to access: /home/user/.library/data/GPX/c3.ini

charely6, 2024-07-11:
just so you know a replicator Mini+ is similar to a replicator 5th gen and uses a completely different thing than this., You will need to use makerbot print 4.3 to communicate with it directly, There is some other projects here on Github for trying to control them but none of them are very easy to use or setup. Sorry.

I have worked on a project called mbotmake that can be tied to orcaslicer or prusaslicer to convert gcode to a makerbot file to print on the 5th gen printers.

wBrhy2, 2024-07-11:
Perfect thank you!